olikraus/ucglib #60: Screen slide down on some ILI9163

dobrMAN, 2015-08-28:
Sorry for my English.
On some displays ILI9163 (I think with black PCB only, not red PCB), Y coordinats shift down on 32 pixels. For fix it, need comment some lines in code ucg_dev_ic_ili9163.c

ucg_int_t ucg_handle_ili9163_l90fx(ucg_t *ucg)
{
  uint8_t c[3];
  ucg_int_t tmp;
  if ( ucg_clip_l90fx(ucg) != 0 )
  {
    switch(ucg->arg.dir)
    {
      case 0: 
//	ucg->arg.pixel.pos.y += 32;
	ucg_com_SendCmdSeq(ucg, ucg_ili9163_set_pos_dir0_seq);	
//	ucg->arg.pixel.pos.y -= 32;
	break;
      case 1: 
//	ucg->arg.pixel.pos.y += 32;
	ucg_com_SendCmdSeq(ucg, ucg_ili9163_set_pos_dir1_seq);	
//	ucg->arg.pixel.pos.y -= 32;
	break;
      case 2: 
	tmp = ucg->arg.pixel.pos.x;
	ucg->arg.pixel.pos.x = 127-tmp;
//	ucg->arg.pixel.pos.y += 32;
	ucg_com_SendCmdSeq(ucg, ucg_ili9163_set_pos_dir2_seq);	
//	ucg->arg.pixel.pos.y -= 32;
	ucg->arg.pixel.pos.x = tmp;
	break;
      case 3: 
      default: 
	tmp = ucg->arg.pixel.pos.y;
	ucg->arg.pixel.pos.y = 127-tmp;
	//ucg->arg.pixel.pos.y += 32;
	ucg_com_SendCmdSeq(ucg, ucg_ili9163_set_pos_dir3_seq);	
	ucg->arg.pixel.pos.y = tmp;
	break;
    }
    c[0] = ucg->arg.pixel.rgb.color[0];
    c[1] = ucg->arg.pixel.rgb.color[1];
    c[2] = ucg->arg.pixel.rgb.color[2];
    ucg_com_SendRepeat3Bytes(ucg, ucg->arg.len, c);
    ucg_com_SetCSLineStatus(ucg, 1);		/* disable chip */
    return 1;
  }
  return 0;
}

Mayby you wrote any definition for ili9163 with black PCB like this - http://www.ebay.com/itm/400684713153?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT

fleinze, 2016-02-13:
I am having the same issue. Maybe you should write a seperate driver for ili9163 without the 32px shift.

FrankX0, 2016-03-21:
I encountered the same problem, but with the red PCB (which looks exactly the same as the black one). Because the ILI9163 can handle different LCD configurations it is more likely that the pixel shifts in the code depends on the actual configuration. So the changes made above might be correct only for the 128x128 configuration.

tpogany, 2016-04-30:
This is because there is a orangeish "red" version, which is same as black, and it is without the 32 shift

olikraus, 2016-05-05:
hmmm... nice finding