olikraus/ucglib #61: Port to ESP8266 (Xtensa)

spapadim, 2015-08-30:
Working on porting this, currently I have it compiling fine, see:
   https://github.com/spapadim/ucglib
(Note: If you want to try this code, Arduino for ESP does not enable -Wl,--gc-sections flag, but just modifying platform.txt to add the necessary compile and link flags seems to have done the trick -- I'll file a separate issue there)

Once I have it tested on an actual device (on it's way from China :), I can create an PR, if you're interested.  If you are, there are a few questions I have about your preferred coding style (and some housecleaning I'd like to do to my changes).

spapadim, 2015-09-13:
Update: tested working on ILI9341.  

Changes commited to my fork.

Also, these short diffs need to be applied to the installed Arduno ESP package:
https://gist.github.com/spapadim/a4bc258df47f00831006

For the record, getting 5.2fps for text, 4.4fps for box, and 4.1fps for gradients.

beegee-tokyo, 2016-05-05:
@spapadim 
I want to use this lib on my ESP8266. Any progress since your last comment? I checked your github repository, but I need some help how to add it to the Arduino IDE libraries.

olikraus, 2016-12-18:
Info via PM on Arduino Forum:
habe es hinbekommen, dass die ucglib am ESP8266 läuft. Bisher keine Auffälligkeiten, hab aber nur drawstring und box sowie frame und setcolor, setfont und setfontmode verwendet.

Dafür musste ich eigentlich nur in der Ucglib.cpp in der Zeile 774 folgendes ändern:

von:
#if defined(__PIC32MX) || defined(__arm__)

auf:
#if defined(__PIC32MX) || defined(__arm__) || defined(ESP8266)

equake, 2017-01-07:
hi @olikraus i've submitted an pull request to the another repo. it's working for me! https://github.com/olikraus/Ucglib_Arduino/pull/3

olikraus, 2017-01-07:
done

closed 2017-01-07 by olikraus