olikraus/ucglib #65: does not compile on teensy 3.X

apleschu, 2016-03-09:
Trying to compile the library on a teensy 3.2 for a ILI9341 320x240 display produces the following errors:

/Users/andy/src/Arduino/libraries/Ucglib/src/Ucglib.cpp: In function 'void ucg_com_arduino_init_8bit(ucg_t*)':
/Users/andy/src/Arduino/libraries/Ucglib/src/Ucglib.cpp:786:20: error: cannot convert 'volatile uint8_t* {aka volatile unsigned char*}' to 'volatile uint32_t* {aka volatile long unsigned int*}' in assignment
u8g_data_port[0] =  portOutputRegister(digitalPinToPort(ucg->pin_list[UCG_PIN_D0]));
^
/Users/andy/src/Arduino/libraries/Ucglib/src/Ucglib.cpp:789:20: error: cannot convert 'volatile uint8_t* {aka volatile unsigned char*}' to 'volatile uint32_t* {aka volatile long unsigned int*}' in assignment
u8g_data_port[1] =  portOutputRegister(digitalPinToPort(ucg->pin_list[UCG_PIN_D1]));
^
/Users/andy/src/Arduino/libraries/Ucglib/src/Ucglib.cpp:792:20: error: cannot convert 'volatile uint8_t* {aka volatile unsigned char*}' to 'volatile uint32_t* {aka volatile long unsigned int*}' in assignment
u8g_data_port[2] =  portOutputRegister(digitalPinToPort(ucg->pin_list[UCG_PIN_D2]));
^
/Users/andy/src/Arduino/libraries/Ucglib/src/Ucglib.cpp:795:20: error: cannot convert 'volatile uint8_t* {aka volatile unsigned char*}' to 'volatile uint32_t* {aka volatile long unsigned int*}' in assignment
u8g_data_port[3] =  portOutputRegister(digitalPinToPort(ucg->pin_list[UCG_PIN_D3]));
^
/Users/andy/src/Arduino/libraries/Ucglib/src/Ucglib.cpp:798:20: error: cannot convert 'volatile uint8_t* {aka volatile unsigned char*}' to 'volatile uint32_t* {aka volatile long unsigned int*}' in assignment
u8g_data_port[4] =  portOutputRegister(digitalPinToPort(ucg->pin_list[UCG_PIN_D4]));
^
/Users/andy/src/Arduino/libraries/Ucglib/src/Ucglib.cpp:801:20: error: cannot convert 'volatile uint8_t* {aka volatile unsigned char*}' to 'volatile uint32_t* {aka volatile long unsigned int*}' in assignment
u8g_data_port[5] =  portOutputRegister(digitalPinToPort(ucg->pin_list[UCG_PIN_D5]));
^
/Users/andy/src/Arduino/libraries/Ucglib/src/Ucglib.cpp:806:22: error: cannot convert 'volatile uint8_t* {aka volatile unsigned char*}' to 'volatile uint32_t* {aka volatile long unsigned int*}' in assignment
u8g_data_port[6] =  portOutputRegister(digitalPinToPort(ucg->pin_list[UCG_PIN_D6]));
^
/Users/andy/src/Arduino/libraries/Ucglib/src/Ucglib.cpp:812:22: error: cannot convert 'volatile uint8_t* {aka volatile unsigned char*}' to 'volatile uint32_t* {aka volatile long unsigned int*}' in assignment
u8g_data_port[7] =  portOutputRegister(digitalPinToPort(ucg->pin_list[UCG_PIN_D7]));
^
/Users/andy/src/Arduino/libraries/Ucglib/src/Ucglib.cpp:816:20: error: cannot convert 'volatile uint8_t* {aka volatile unsigned char*}' to 'volatile uint32_t* {aka volatile long unsigned int*}' in assignment
u8g_data_port[8] =  portOutputRegister(digitalPinToPort(ucg->pin_list[UCG_PIN_WR]));
^
/Users/andy/src/Arduino/libraries/Ucglib/src/Ucglib.cpp: In function 'void ucg_com_arduino_send_8bit(ucg_t*, uint8_t)':
/Users/andy/src/Arduino/libraries/Ucglib/src/Ucglib.cpp:825:3: error: '__NOP' was not declared in this scope
__NOP;
^
exit status 1
Error compiling.

jgilbert20, 2016-08-01:
Same issue here. Using 1.29 Teensy + last release.

emilien-g, 2016-08-26:
Same issue with Teensyduino1.29  and Arduino 1.6.9

olikraus, 2017-01-07:
I need more information here. __NOP is only used if __arm__ is defined. But then, __NOP should be available.

Jorgen-VikingGod, 2017-04-24:
I had the same issues on my Teensy 3.1 setup.
Did some small changes to support the Teensy 3.1/3.2
https://github.com/olikraus/Ucglib_Arduino/pull/7

mamama1, 2018-07-15:
Hi

same here for the Teensy 3.5.
@Jorgen-VikingGod your fix works for me, if I change __MK20DX256__ to __MK64FX512__, which is the Teensy 3.5 CPU.

Is this going to be fixed at some point?

Thanks!

dentaku-web, 2018-09-21:
Well, it seems I'm not the only person having the problem so that's good I guess but not good that it's been happening since 2016.
Arduino IDE version 1.8.5 and Ucglib version 1.5.2.

C:\Users\dentaku\Documents\Arduino\libraries\Ucglib\src\Ucglib.cpp: In function 'void ucg_com_arduino_init_8bit(ucg_t*)':

C:\Users\dentaku\Documents\Arduino\libraries\Ucglib\src\Ucglib.cpp:795:20: error: cannot convert 'volatile uint8_t* {aka volatile unsigned char*}' to 'volatile uint32_t* {aka volatile long unsigned int*}' in assignment

   u8g_data_port[0] =  portOutputRegister(digitalPinToPort(ucg->pin_list[UCG_PIN_D0]));

                    ^

C:\Users\dentaku\Documents\Arduino\libraries\Ucglib\src\Ucglib.cpp:798:20: error: cannot convert 'volatile uint8_t* {aka volatile unsigned char*}' to 'volatile uint32_t* {aka volatile long unsigned int*}' in assignment

   u8g_data_port[1] =  portOutputRegister(digitalPinToPort(ucg->pin_list[UCG_PIN_D1]));

                    ^

C:\Users\dentaku\Documents\Arduino\libraries\Ucglib\src\Ucglib.cpp:801:20: error: cannot convert 'volatile uint8_t* {aka volatile unsigned char*}' to 'volatile uint32_t* {aka volatile long unsigned int*}' in assignment

   u8g_data_port[2] =  portOutputRegister(digitalPinToPort(ucg->pin_list[UCG_PIN_D2]));

                    ^

C:\Users\dentaku\Documents\Arduino\libraries\Ucglib\src\Ucglib.cpp:804:20: error: cannot convert 'volatile uint8_t* {aka volatile unsigned char*}' to 'volatile uint32_t* {aka volatile long unsigned int*}' in assignment

   u8g_data_port[3] =  portOutputRegister(digitalPinToPort(ucg->pin_list[UCG_PIN_D3]));

                    ^

C:\Users\dentaku\Documents\Arduino\libraries\Ucglib\src\Ucglib.cpp:807:20: error: cannot convert 'volatile uint8_t* {aka volatile unsigned char*}' to 'volatile uint32_t* {aka volatile long unsigned int*}' in assignment

   u8g_data_port[4] =  portOutputRegister(digitalPinToPort(ucg->pin_list[UCG_PIN_D4]));

                    ^

C:\Users\dentaku\Documents\Arduino\libraries\Ucglib\src\Ucglib.cpp:810:20: error: cannot convert 'volatile uint8_t* {aka volatile unsigned char*}' to 'volatile uint32_t* {aka volatile long unsigned int*}' in assignment

   u8g_data_port[5] =  portOutputRegister(digitalPinToPort(ucg->pin_list[UCG_PIN_D5]));

                    ^

C:\Users\dentaku\Documents\Arduino\libraries\Ucglib\src\Ucglib.cpp:815:22: error: cannot convert 'volatile uint8_t* {aka volatile unsigned char*}' to 'volatile uint32_t* {aka volatile long unsigned int*}' in assignment

     u8g_data_port[6] =  portOutputRegister(digitalPinToPort(ucg->pin_list[UCG_PIN_D6]));

                      ^

C:\Users\dentaku\Documents\Arduino\libraries\Ucglib\src\Ucglib.cpp:821:22: error: cannot convert 'volatile uint8_t* {aka volatile unsigned char*}' to 'volatile uint32_t* {aka volatile long unsigned int*}' in assignment

     u8g_data_port[7] =  portOutputRegister(digitalPinToPort(ucg->pin_list[UCG_PIN_D7]));

                      ^

C:\Users\dentaku\Documents\Arduino\libraries\Ucglib\src\Ucglib.cpp:825:20: error: cannot convert 'volatile uint8_t* {aka volatile unsigned char*}' to 'volatile uint32_t* {aka volatile long unsigned int*}' in assignment

   u8g_data_port[8] =  portOutputRegister(digitalPinToPort(ucg->pin_list[UCG_PIN_WR]));

                    ^

C:\Users\dentaku\Documents\Arduino\libraries\Ucglib\src\Ucglib.cpp: In function 'void ucg_com_arduino_send_8bit(ucg_t*, uint8_t)':

C:\Users\dentaku\Documents\Arduino\libraries\Ucglib\src\Ucglib.cpp:834:3: error: '__NOP' was not declared in this scope

   __NOP;

   ^

Using library SPI at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SPI 
Using library Ucglib at version 1.5.2 in folder: C:\Users\dentaku\Documents\Arduino\libraries\Ucglib 
Error compiling for board Teensy 3.2 / 3.1.

MattMatic, 2018-09-21:
Actually, having got ucglib working I checked out all the ILI9341_t3 variants again. In the end I went for blackketter's version - https://github.com/blackketter/ILI9341_t3  - it's blazingly fast and has a reasonable amount of font handling to support right justified text etc.

It wasn't that hard to switch over in the end.

Sorry Oli.

MattMatic, 2018-09-24:
Good to hear you're still around Oli! The U8g2 library is excellent - keep up the good work on that ;)