olikraus/ucglib #66: Draw glyphs or images without a font

hspeedy, 2016-03-13:
Is it possible to draw images without creating a font?
I tried to create a font using Fony and your bdf converter.
This didn't work for me because something is missing.
The created file was always the same. Is there something
I need to know? I'm sure this would be a good way but
there should be more information in your wiki about doing
this.

Greetings Heiko

hspeedy, 2016-03-13:
Okay, now it works. Seems as I have to start at 0x20 with this font. My last tries where from 0x00 to 0x0B.

closed 2016-03-13 by hspeedy

TCB13, 2023-01-20:
@hspeedy can you share how you created the font and made it work?

hspeedy, 2023-01-21:
i have used a tool to create fonts by import or drawing the characters with pixels. i don't remember the name. seemed to be an old win32 tool. at the end there have been different export option like ttf or a code fornat consisting of a large byte array to use in code.

TCB13, 2023-01-21:
@hspeedy thank you for taking the time to answer me. If you've ever come across the tool's name let me know :)

Meanwhile I'll be using the Adafruit GFX that for all its defects makes it way easier to print glyphs / convert fonts etc.