olikraus/ucglib #69: Port to ESP32 (Xtensa)

pcbreflux, 2016-11-20:
Working on a Port to ESP32
see my suggestion
https://github.com/pcbreflux/espressif/tree/master/esp32/arduino/libraries/Ucglib/src/Ucglib.cpp

Tested on a ESP32 inside Arduino IDE with ILI9341 using Box3D, Font, Text and Color

equake, 2017-01-07:
@pcbreflux I'm using this lib with ESP8266. My change is smaller. Take a look: https://github.com/olikraus/Ucglib_Arduino/pull/3

olikraus, 2017-01-07:
Will this also work for the ESP32?
#if defined(__PIC32MX) || defined(__arm__) || defined(ESP8266) || defined(ESP_PLATFORM)

pcbreflux, 2017-01-07:
hi olikraus,

could try it on monday and give feedback.

have a nice weekend
pcbreflux

<https://github.com/olikraus/ucglib/issues/69#issuecomment-271086786>, or mute
the thread

Dennis650, 2017-01-07:
Hi,

i tried version 1.4 with the esp32, but it didn't compile in my inviremont:

```
/Users/dennis/Documents/Arduino/libraries/Ucglib/src/Ucglib.cpp:786:20: error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
   u8g_data_port[0] =  portOutputRegister(digitalPinToPort(ucg->pin_list[UCG_PIN_D0]));

```

I have to use the fix of pcbreflux again.

Greetings
Dennis

olikraus, 2017-01-08:
ok, the goal was to make it work for "ESP8266".
ESP32 was more an extra wish.
Result now is, that ESP32 requires 32 bit arrays.

Jorgen-VikingGod, 2017-01-08:
I also used the hack of @pcbreflux to run it on my ESP32 devboard.

olikraus, 2017-01-08:
Are there any instructions how to setup Arduiono IDE for ESP32?

Dennis650, 2017-01-08:
Hi,

yes: https://github.com/espressif/arduino-esp32

Greetings
Dennis

pcbreflux, 2017-01-09:
Hi Olikraus,

test the Patch from Marcos A Sobrino.

Using
- Arduino 1.6.13 linux 64Bit and Arduino 1.8 linux 64Bit
- Arduino core for ESP32 form https://github.com/espressif/arduino-esp32
(git version 7cef2e2)
- ucglib Version 1.4.0 with patch on Line 774
- ILI9341 with SWSPI and HWSPI
- HelloWorld.ino, GraphicsTest.ini (without line 561 set_clip_range()),
Box3D.ino

Settings
//Ucglib_ILI9341_18x240x320_SWSPI ucg(/*sclk=*/ 18, /*data=*/ 23, /*cd=*/
16 , /*cs=*/ 5, /*reset=*/ 17);
//Ucglib_ILI9341_18x240x320_HWSPI ucg(/*cd=*/ 16 , /*cs=*/ 5, /*reset=*/
17);

Works fine for me (only some warnings), so respect to Marcos for the small
solution.

But only one oddness not related with the patch: on the ESP32 HWSPI works
much slower then SWSPI (maybe 2 - 3 times faster).

Have a nice day
pcbreflux

Jorgen-VikingGod, 2017-01-09:
Would be great to have full speed support by HWSPI
fill screen and changes are flicker too much...

Dennis650, 2017-01-09:
Hi,

@Jorgen-VikingGod 
@pcbreflux 
can you please post a pin mapping for HWSPI? I have a IL9431 like seen in the video from @pcbreflux.
I tried it, but it won't work.
Do you have touch enabled? 

Greetings
Dennis

pcbreflux, 2017-01-09:
Hi Dennis,

using this pinmap https://github.com/espressif/arduino-esp32/blob/
master/doc/esp32_pinmap.png

No touch. Using the same pins for HWSPI and SSSPI.

Settings
//Ucglib_ILI9341_18x240x320_SWSPI ucg(/*sclk=*/ 18, /*data=*/ 23, /*cd=*/
16 , /*cs=*/ 5, /*reset=*/ 17);
//Ucglib_ILI9341_18x240x320_HWSPI ucg(/*cd=*/ 16 , /*cs=*/ 5, /*reset=*/
17);

Greetings
pcbreflux

olikraus, 2017-01-09:
ok,i guess i will order such an ESP32 for further tests. Any suggestion what i can buy from ebay.com?

pcbreflux, 2017-01-10:
Hi,

I tried ebay (eckstein in germany) and you can look at analoglamb.com . For
me aliexpress is not reliable today. ordered from differend sellers but
never came and have to refund after 6 to 8 weeks.

have a nice day
pcbreflux

Dennis650, 2017-01-10:
Hi, 

@pcbreflux HWSPI is working now, but it is realy slow, as you said.

@olikraus: i ordered a SparkFun ESP32 Thing (from eckstein germany, too).

Greetings
Dennis

olikraus, 2017-01-10:
so you mean this
https://eckstein-shop.de/SparkFun-ESP32-Thing-Espressif-Inside-WLAN-BLE-IoT-Modul
or this
http://www.exp-tech.de/sparkfun-esp32-thing
correct?
Maybe i get this thing from exp-tech... I had previously ordered from exp-tech already. :-p

Dennis650, 2017-01-10:
Hi,

yes. And thanks for your great work on this lib.

Greetings
Dennis

Jorgen-VikingGod, 2017-01-10:
@olikraus I orded mine devboard in this shop:
http://www.watterott.com/de/ESP-WROOM32-Breakout

olikraus, 2017-01-10:
sagt mal... das sind ja alles Deutsche shops...
ok, thanks for all the answer, i guess i will order the thing from exp-tech

Dennis650, 2017-01-10:
:-)

olikraus, 2017-01-10:
ordered...

olikraus, 2017-01-14:
When i try to upload, i get the followig error:
A fatal error occurred: Failed to connect to ESP32: Invalid head of packet ('r')
any idea?

Arduino IDE: 1.8.1
Board: ESP32 Dev Module

olikraus, 2017-01-14:
ESP32

defines by the compiler
#define __XTENSA_EL__ 1
#define __xtensa__ 1
#define __XTENSA__ 1
#define __XTENSA_WINDOWED_ABI__ 1

defines by Arduino IDE:
ESP_PLATFORM

pcbreflux, 2017-01-14:
hi, before uploading you have to bring the esp32 to donwload mode gpio0 to
low (long press) and reset (short press) and sometimes make shure to
disconnect any serial monitor. hope this helps.

Greetings
pcbreflux

olikraus, 2017-01-14:
Thanks, i tried this and i think some communication starts (blue leds becomes dim and i see some flicker during the connect attempt)
Actually i try this:
~$ "python" "/home/kraus/Arduino/hardware/espressif/esp32/tools/esptool.py" --chip esp32 chip_id 
esptool.py v2.0-dev
Connecting...

A fatal error occurred: Failed to connect to ESP32: Invalid head of packet ('r')
~$

olikraus, 2017-01-14:
hmmm... whatever i try it is not working.

pcbreflux, 2017-01-14:
Hi,

have you try connecting with the serial monitor (115200 baud) and just
watch the debug output after reset? then go to download mode ...... read
somthing like 'waiting for download ....'?

olikraus, 2017-01-14:
I have this msg after reboot:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
```

then, after the upload attempt, i get the same error with this output on the serial monitor:

```
frc2_timer_task_hdl:3ffc53ac, prio:22, stack:2048
tcpip_task_hdlxxx : 3ffc6624, prio:18,stack:2048
phy_version: 123, Sep 13 2016, 20:01:58, 1
pp_task_hdl : 3ffc9e5c, prio:23, stack:8192
mode : sta(24:0a:c4:04:b8:f0)
```

olikraus, 2017-01-14:
I had a look at your code again:
https://github.com/pcbreflux/espressif/blob/master/esp32/arduino/libraries/Ucglib/src/Ucglib.cpp

The problem is, that i du not understand your code. Why is port = 0?
The ESP32 has at least two port registers. 

Anyhow, i give up here. Maybe once this ESP32 is more stable and reliable...

pcbreflux, 2017-01-14:
hope you esp32 is not broken. but after the "waiting for download" can you
try to close the serial monitor before uploading? what board, baudrate and
usbport you are using?

olikraus, 2017-01-14:
Closing the serial monitor does not make a difference. Still, it does not work.
I do not think its broken, i just think the the upload tool does not work.
My ESP thing has a very new version (13 Sep 2016), so maybe it is because of this.

Dennis650, 2017-01-14:
Hi,

i have installed Arduino 1.8 on a Mac.  I just downloaded an example and uploaded it to the Esp32 through the Arduino IDE by clicking on "Hochladen". There is no need to pull GPIO0 to low. 

Here is a guide: https://learn.sparkfun.com/tutorials/esp32-thing-hookup-guide#installing-the-esp32-arduino-core

Greetings
Dennis

olikraus, 2017-01-14:
I did follow exactly these instructions. I have used two different linux machines. Always the same error.
Ok, i have not yet used Mac or Windows.

Dennis650, 2017-01-27:
Hi,

there were a lot of work the last days on arduino for esp32. Maybe you can try again and post an issue if it doesn't work?

But there is also an new ticket for slow hardware SPI: https://github.com/espressif/arduino-esp32/issues/149

Greetings
Dennis

olikraus, 2017-12-12:
should work with 1.5.x

olikraus, 2017-12-16:
at least sw spi should work now with ESP32.

closed 2017-12-16 by olikraus