olikraus/ucglib #72: ILI9341 countdown display issues ..  HELP..

svishnu06, 2017-02-17:
i am testing with stm32f103c8t6 and ili9341 combination, problem is when trying to display a counter (numbers) , for example - to display countdown number 110 to 1 , after "100" display shows "990" , not "99" ,last zero is still there. after "10" , "90" not '9'.. at last "100" instead of '1', then ''000'' for '0' , how to solve this problem ?
a video link.. https://youtu.be/mer0_pwQtsE
help me to solve this .

FrankX0, 2017-02-19:
You are printing left-aligned. This means that if you go from 3 to 2 digits (100 to 99) the last digit (0) remains on the display. Same happens when counting from 10 to 1.
To resolve you could clear the counter area before updating or using formatting to always print 3 digits.

svishnu06, 2017-02-20:
I am just a student .. here is my testing code .. (output is here -https://youtu.be/mer0_pwQtsE)..  

how to solve it, help me  ?
 

#include <SPI.h>
#include <Ucglib.h>
Ucglib_ILI9341_18x240x320_HWSPI ucg(/*cd=*/ PA2 , /*cs=*/ PA0, /*reset=*/ PA1);

int counter = 120;

void setup() {
  Serial.begin(115200);
  delay(1000);
  ucg.begin(UCG_FONT_MODE_TRANSPARENT);
  ucg.clearScreen();
}

void loop() {
  counter--;

  ucg.setRotate270();
  ucg.setFontMode(UCG_FONT_MODE_SOLID);
  ucg.setFont(ucg_font_inb33_mn);
  ucg.setColor(255, 255, 255);
  ucg.setPrintPos(45, 200);
  ucg.println( counter);

  delay(200);

}

FrankX0, 2017-02-20:
Depends on what you want to achieve there are multiple solutions:

1. Use clearScreen before printing new value
2. Perform a local clear of the screen, using drawBox, before printing the new value
3. Convert your counter to a string by using formatting to always get 3 digits (100, 099, ..., 010, 009, etc.), and using drawString for printing
4. Convert your counter to a string with spaces added in front, get the width of the resulting string (getStrWidth ) and subtract from your x-position, and use this new position: this will result in right-alignment and with "old"  digits cleared by the added spaces.

Give it a try, start experimenting.

v8mgb, 2017-02-20:
Add a space If the number is less then 99

Jim Miller
269-923-6231 office
269-208-8391 cell