olikraus/ucglib #78: ST7735 with esp32

amardhore, 2017-07-25:
Great job writing all these display libraries. I tried the ST7735 libraries with ESP32 (Arduino IDE). It kind of works. 
I was very much interested to make st7735 work with espressif SDK. So far I had no luck. If anyone did it, I would appreciate the help.

olikraus, 2017-12-12:
... working on this.... 
It should work now on ESP32

amardhore, 2017-12-12:
@olikraus I made little changes to the library and  I am running ST7735 with ESP32 sdk. I am still trying to enhance few functions like clearscreen(). Thanks again.
--Amar

olikraus, 2017-12-12:
I tested several other displays with ESP32, seems to work as of now.
SPI not yet tested.

amardhore, 2017-12-15:
I had issues with the software SPI where display will stuck at a white screen after few hours of run. I moved to hardware SPI, display works flawlessly. Only issue is, the ucg_ClearScreen() is very slow. I wonder if there is any clearscreen command that driver accepts, Instead of filling a screen with a black color .

olikraus, 2017-12-16:
To my knowledge there is no such controller command for clear screen....
I will close this, feel free to open another issue for any speed related problems.

closed 2017-12-16 by olikraus