olikraus/ucglib #79: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'volatile uint8_t*

nabulsi, 2017-10-12:
Hi,

I have an ILI9341 display. I am using the latest Arduino (1.8.5) and latest Ucglib (1.4.0). I have tried on 2 different machines, MAC & Windows, but still getting the same error:
`/Users/xxxx/Documents/Arduino/libraries/Ucglib/src/Ucglib_orig.cpp: In function 'void ucg_com_arduino_init_8bit(ucg_t*)':
/Users/xxxxx/Documents/Arduino/libraries/Ucglib/src/Ucglib_orig.cpp:786:20: error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
   u8g_data_port[0] =  portOutputRegister(digitalPinToPort(ucg->pin_list[UCG_PIN_D0]));`

there are in total 9 errors like this (i.e. from u8g_data_port[0] to [8]). In addition, in both machines I see the library under "Incompatible" when going to "File --> Examples".

This is my constructor:
Ucglib_ILI9341_18x240x320_SWSPI ucg(/*sclk=*/ 18, /*data=*/ 23, /*cd=*/ 13 , /*cs=*/ 5, /*reset=*/ 19);

Thanks

nabulsi, 2017-10-24:
Anyone is able to shed some light here please?

Thanks

olikraus, 2017-12-12:
which board? 
I think it will be fixed for 1.5.x

nabulsi, 2017-12-13:
2 boards:
   ESP32-DevKitC & SparkFun ESP32

olikraus, 2017-12-13:
Yes, this is fixed. I had the same issue with the ESP32 boards.

closed 2017-12-16 by olikraus