olikraus/ucglib #87: Support for STM32

dasaki, 2018-03-21:
In order to add support for stm32duino platforms the following definition is needed:

#if defined(__arm__)
#ifndef __NOP
#define __NOP __asm volatile ("nop")
#endif
#endif

Adding the definition to the file Ucglib.c worked with a maple mini clone and a ST7735 TFT:
Ucglib_ST7735_18x128x160_HWSPI ucg(/*cd=*/ PA2 , /*cs=*/ PA4, /*reset=*/ PA3);

mithat, 2018-11-17:
Confirming that the definition works with `Ucglib_ILI9341_18x240x320_HWSPI()` on a [Blue Pill](https://wiki.stm32duino.com/index.php?title=Blue_Pill) as well.

konosubakonoakua, 2019-03-26:
NOT WORK FOR with `Ucglib_ST7735_18x128x160_HWSPI ucg(/*cd=*/ PA1, /*cs=*/ PA0, /*reset=*/ PA2);` on a [Blue Pill](https://wiki.stm32duino.com/index.php?title=Blue_Pill).
BUT WORKS just with `#define __NOP __ASM volatile ("nop");` added to `Ucglib.cpp` : line 43

dan69p, 2019-12-01:
Works on China board with stm32f103 **rc** (diymore)

Nimajine, 2021-02-20:
Hi, for me WORKS just with `#define __NOP __asm volatile ("nop");` added to Ucglib.cpp : line 43
and in IDE I use: `Ucglib_ST7735_18x128x160_HWSPI ucg(/*cd=*/PB1, /*cs=*/ PB12, /*reset=*/ PA4);`
(STM32103FC with STM32duino core 2021.2.19)