olikraus/ucglib #93: Feature: Support for ST7735S 80x160 .96"

tablatronix, 2018-07-26:


tablatronix, 2018-07-26:
[image]
[image]

sirpete83, 2018-12-23:
Hey,
I received 2 of this displays. Found out how it works, will present a workaround for this display type. have it functional on my table, but i have to do some tests.

tablatronix, 2018-12-23:
I have this working with bodmer espi library which had my pr merged, I was not familiar with the code changes needed for this, but you can check my pr for refs.

https://github.com/Bodmer/TFT_eSPI/pull/179/files

Bgr inverted, offsets in pr