olikraus/ucglib #95: SSD1351 not updating without using ucg.begin()

NSchnitzler, 2018-09-12:
Hi,

i am making my first steps with a SSD1351 128x128 RGB OLED in combination with a Arduino DUE. My problem is, that the Display is only updatingafter using the ucg.begin(..) function at the beginning of every update like shown in the small snippet below. After commenting out this line, the display keeps the old content.

`void updateScreen(void) {
  ucg.begin(UCG_FONT_MODE_TRANSPARENT); <--- Display not updating without this
  ucg.clearScreen();
  ucg.setFont(ucg_font_helvB10_hr);
  ucg.setPrintPos(0, 40);
  ucg.print(String(rnd));
  ucg.setPrintPos(0, 60);
  ucg.print(frequency);
}
`

The init look like:
`
Ucglib_SSD1351_18x128x128_HWSPI ucg(/*cd=*/ 9, /*cs=*/ 10, /*reset=*/ 11);
// **********  DISPLAY INIT ***************
uint8_t z = 127;  // start value
uint32_t lcg_rnd(void) {
  z = (uint8_t)((uint16_t)65 * (uint16_t)z + (uint16_t)17);
  return (uint32_t)z;
}

setup:
  ucg.begin(UCG_FONT_MODE_TRANSPARENT);
  ucg.setFont(ucg_font_helvB14_hr);
  ucg.clearScreen();
`

MattMatic, 2018-09-14:
I have a similar issue that's related. With ILI9341 and XPT2046, the SPI bus goes down to the lowest common denominator (2MHz for touch). Actually, I think it's the last SPISettings that gets called.

Unfortunately it looks like UCGLIB isn't using SPI.begin / .beginTransaction / .end for each transfer - but assuming it has exclusive use of the SPI bus :-(

(FWIW, I have an Arduino project on Teensy 3.6 that's built under Segger's Embedded Studio and J-Link programmer - which makes debugging this much easier)

Matt

NSchnitzler, 2018-10-02:
I switched to SW-SPI and now both devices were working as expected.