olikraus/ucglib #97: Conflict with SD card module?

jagganath69, 2018-09-16:
My ST7735 works perfectly. So does my SD module. But when I use them together on my Mega2560, the display blanks out and the SD init() says error on initialization. (They share the usual MISO, CLK, etc.
They have individual SS pins assigned)

When I pull the MISO pin 50 off one, the other works OK. Although I don't know if it's the CS line that has timing problems (I'm not much into hardware)

gsfci, 2019-01-02:
I have almost the same issue. But I use ESP32 MCU and ILI9341 display. SD request works just fine, until I use the ucg.begin() function. After that, any SD request (including SD.begin, if I didn't call it earlier) freezes the board. That's not a hardware issue - Adafruit's ILI9341 library works as it should with the SD library.

mateusz-kusmierz, 2019-12-02:
I have the same problem. Did anyone managed how to overcome it?

OmerFlame, 2020-05-27:
There is a way to mitigate this issue, but I am facing a dilemma when it comes to TFT displays with SD card backplanes.

You could use a software SPI solution (faking a SPI bus using a library called SoftSPI), with it you can set up your OWN MOSI, MISO, CS and SCK pins without interrupting the actual hardware SPI bus. I had to do this because I use a ST7735 and 128x160 display combo with a SD backplane made by Adafruit, but I can't theoretically figure out how I would use Ucglib AND the Adafruit SPI SD card library at the same time...