olikraus/ucglib #100: ili9341 get cropped with m5stack esp32 module

pastukhov, 2018-11-08:
Hi all!
I'm trying to get working ucglib with [m5stack](http://www.m5stack.com/assets/docs/) module that included ili9341 screen and get output cropped by 240x240 pixels
See attached [Video](https://www.youtube.com/watch?v=ctrHVP_pt-o)
Any advice?

AnHardt, 2018-11-08:
Rotate 90°.

pastukhov, 2018-11-09:
Same, but rotated.
[image]

f4grx, 2021-09-30:
ucg does not work with m5stack. I did research and got it working.

I did that by adding a new display variant for the ili9341, rotation is not enough.

f4grx, 2021-09-30:
Here is my branch with proper support. it is not against this repository but against the _arduino version.

https://github.com/f4grx/Ucglib_Arduino/tree/m5stack/

f4grx, 2021-09-30:
In fact there is even a pull request since january but it was not merged yet :(

https://github.com/olikraus/Ucglib_Arduino/pull/23