olikraus/ucglib #106: Problem with 1,8" 128x160 TFT Display ST7735 shifted screen

karawin, 2018-11-27:
Some device are ok, some are shifted from one or two pixels in axis x and y
What can be modified in ucg_dev_ic_st7735.c, ucg_dev_tft_128x160_st7735.c to change the offset?
Thanks

[image]

sirpete83, 2018-11-27:
Hey karawin,
I got the same problem. see #105 . I don't believe that this is a problem with the code or the library. I think it is a problem with the display itself. I ordered a new one from an other distributor. the faulty one I got on aliexpess, only have one.

karawin, 2018-11-28:
I think we are in -132 (RGB) x 162
(GM[2:0]= ”000”, DDRAM: 132 x 18-bits x 162) resolution.

sirpete83, 2018-11-28:
Hey karawin, 
you are right, that modules are in GM 000 Mode, Im working on this! but it can take so time.

karawin, 2018-12-01:
Modified ucg_dev_tft_128x160_st7735.c
```
/*

  ucg_dev_tft_128x160_W_st7735.c
  
  ST7735 with 4-Wire SPI (SCK, SDI, CS, D/C and optional reset)

  Universal uC Color Graphics Library
  
  Copyright (c) 2014, olikraus@gmail.com
  All rights reserved.

  Redistribution and use in source and binary forms, with or without modification, 
  are permitted provided that the following conditions are met:

  * Redistributions of source code must retain the above copyright notice, this list 
    of conditions and the following disclaimer.
    
  * Redistributions in binary form must reproduce the above copyright notice, this 
    list of conditions and the following disclaimer in the documentation and/or other 
    materials provided with the distribution.

  THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND 
  CONTRIBUTORS "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, 
  INCLUDING, BUT NOT LIMITED TO, THE IMPLIED WARRANTIES OF 
  MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE 
  DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR 
  CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, 
  SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT 
  NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; 
  LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER 
  CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, 
  STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) 
  ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF 
  ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.  

*/

#include "ucg.h"

//static const uint8_t ucg_dev_ssd1351_128x128_init_seq[] PROGMEM = {
static const ucg_pgm_uint8_t ucg_tft_128x160_W_st7735_init_seq[] = {
  UCG_CFG_CD(0,1),				/* DC=0 for command mode, DC=1 for data and args */
  UCG_RST(1),					
  UCG_CS(1),					/* disable chip */
  UCG_DLY_MS(5),
  UCG_RST(0),					
  UCG_DLY_MS(5),
  UCG_RST(1),
  UCG_DLY_MS(50),
  UCG_CS(0),					/* enable chip */
  
  UCG_C10(0x011),				/* sleep out */
  UCG_DLY_MS(10),
  //UCG_C10(0x038),				/* idle mode off */
  UCG_C10(0x013),				/* normal display on */

  

  UCG_C10(0x20), 				/* not inverted */
  //UCG_C10(0x21), 				/* inverted */

  UCG_C11(0x03a, 0x006), 		/* set pixel format to 18 bit */
 //UCG_C11(0x03a, 0x005), 		/* set pixel format to 16 bit */

  //UCG_C12(0x0b1, 0x000, 0x01b), 	/* frame rate control (POR values) */
  //UCG_C10(0x28), 				/* display off */
  //UCG_C11(0x0bf, 0x003), 		/* backlight control 8 */
  UCG_C10(0x029), 				/* display on */
  //UCG_C11(0x051, 0x07f), 		/* brightness */
  //UCG_C11(0x053, 0x02c), 		/* control brightness */
  //UCG_C10(0x028), 				/* display off */

  UCG_C11( 0x036, 0x000),		/* memory control */
//  UCG_C11( 0x036, 0x0C0),		/* memory control */
  
//  UCG_C14(  0x02a, 0x000, 0x000, 0x000, 0x07f),              /* Horizontal GRAM Address Set */
//  UCG_C14(  0x02b, 0x000, 0x000, 0x000, 0x09f),              /* Vertical GRAM Address Set */
  UCG_C14(  0x02a, 0x000, 0x002, 0x000, 0x081),              /* Horizontal GRAM Address Set */
  UCG_C14(  0x02b, 0x000, 0x001, 0x000, 0x0a0),              /* Vertical GRAM Address Set */
  UCG_C10(  0x02c),               /* Write Data to GRAM */

  
  UCG_CS(1),					/* disable chip */
  UCG_END(),					/* end of sequence */
};

ucg_int_t ucg_dev_st7735_18x128x160_W(ucg_t *ucg, ucg_int_t msg, void *data)
{
  switch(msg)
  {
    case UCG_MSG_DEV_POWER_UP:
      /* 1. Call to the controller procedures to setup the com interface */
      if ( ucg_dev_ic_st7735W_18(ucg, msg, data) == 0 )
	return 0;

      /* 2. Send specific init sequence for this display module */
      ucg_com_SendCmdSeq(ucg, ucg_tft_128x160_W_st7735_init_seq);
      
      return 1;
      
    case UCG_MSG_DEV_POWER_DOWN:
      /* let do power down by the conroller procedures */
      return ucg_dev_ic_st7735W_18(ucg, msg, data);  
    
    case UCG_MSG_GET_DIMENSION:
		return ucg_dev_ic_st7735W_18(ucg, msg, data);
//      ((ucg_wh_t *)data)->w = 128;
//      ((ucg_wh_t *)data)->h = 160;
//      return 1;
  }

  /* all other messages are handled by the controller procedures */
  return ucg_dev_ic_st7735W_18(ucg, msg, data);  
}
```

karawin, 2018-12-01:
Modified ucg_dev_ic_st7735.c in CSRC
```
/*

  ucg_dev_ic_st7735.c
  
  Specific code for the st7735 controller (TFT displays)

  Universal uC Color Graphics Library
  
  Copyright (c) 2014, olikraus@gmail.com
  All rights reserved.

  Redistribution and use in source and binary forms, with or without modification, 
  are permitted provided that the following conditions are met:

  * Redistributions of source code must retain the above copyright notice, this list 
    of conditions and the following disclaimer.
    
  * Redistributions in binary form must reproduce the above copyright notice, this 
    list of conditions and the following disclaimer in the documentation and/or other 
    materials provided with the distribution.

  THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND 
  CONTRIBUTORS "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, 
  INCLUDING, BUT NOT LIMITED TO, THE IMPLIED WARRANTIES OF 
  MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE 
  DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR 
  CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, 
  SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT 
  NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; 
  LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER 
  CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, 
  STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) 
  ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF 
  ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.  

*/

#include "ucg.h"
#define IX	2
#define IY 1
#define WIDTH	160
#define HEIGHT 	128

const ucg_pgm_uint8_t ucg_st7735_set_pos_seq[] = 
{
  UCG_CS(0),					/* enable chip */
  UCG_C11( 0x036, 0x000),
  UCG_C10(0x02a),	UCG_VARX(0,0x00, 0), UCG_VARX(0,0x0ff, 0), UCG_A2(0x000, 0x07f),					/* set x position */
  UCG_C10(0x02b),	UCG_VARY(0,0x00, 0), UCG_VARY(0,0x0ff, 0), UCG_A2(0x000, 0x09f),		/* set y position */
  UCG_C10(0x02c),							/* write to RAM */
  UCG_DATA(),								/* change to data mode */
  UCG_END()
};

const ucg_pgm_uint8_t ucg_st7735_set_pos_dir0_seq[] = 
{
  UCG_CS(0),					/* enable chip */
  
  /* 0x000 horizontal increment (dir = 0) */
  /* 0x000 vertical increment (dir = 1) */
  /* 0x040 horizontal deccrement (dir = 2) */
  /* 0x080 vertical deccrement (dir = 3) */
  UCG_C11( 0x036, 0x000),
  UCG_C10(0x02a),	UCG_VARX(0,0x00, 0), UCG_VARX(0,0x0ff, 0), UCG_A2(0x000, 0x07f),					/* set x position */
  UCG_C10(0x02b),	UCG_VARY(0,0x00, 0), UCG_VARY(0,0x0ff, 0), UCG_A2(0x000, 0x09f),		/* set y position */

  UCG_C10(0x02c),							/* write to RAM */
  UCG_DATA(),								/* change to data mode */
[... 315 lines cut ...]
      return ucg->ext_cb(ucg, msg, data);
    */
  }
  	ucg->arg.pixel.pos.x -= IX;
	ucg->arg.pixel.pos.y -= IY;	  
  return ucg_dev_default_cb(ucg, msg, data);  
}

ucg_int_t ucg_ext_st7735_18(ucg_t *ucg, ucg_int_t msg, void *data)
{
	ucg->arg.pixel.pos.x += IX;
	ucg->arg.pixel.pos.y += IY;
  switch(msg)
  {
    case UCG_MSG_DRAW_L90SE:
      //ucg_handle_l90se(ucg, ucg_dev_ic_st7735_18);
      ucg_handle_st7735_l90se(ucg);
      break;
  }
 	ucg->arg.pixel.pos.x -= IX;
	ucg->arg.pixel.pos.y -= IY;	  
 
  return 1;
}
```

sirpete83, 2018-12-02:
Hey, 
I also have an solution, I think in yours there are to much OFFSET corrections. can be an workaround for the moment, but i would be only functional for this case. 
Maybe I would have the possibility to commit tomorrow.

karawin, 2018-12-02:
Would be great. My answer is wrong anyway.

sirpete83, 2018-12-10:
hey karawin, hey oliver, 
in my fork you can find a test branch for the GM00 Mode Display please test it, it works for my display. maybe we can marge it, with some defines to select the "right" mode fot the different display around the world.

karawin, 2018-12-12:
I took some time to adapt to my code, but it seems to works well.
Congrat!

IAmOrion, 2019-03-25:
@sirpete83 Just FYI, I just came across this thread after trying out a WaveShare 1.8" 128x160 TFT (https://www.waveshare.com/wiki/1.8inch_LCD_Module) ... I had the same issues as the OP - your "offset_test1_st7735" branch WORKS for me.  Thought I'd let you know in case you want to add the Waveshare variant to a list of devices that your fork fixes 👍

a0908896067, 2020-03-30:
> Some device are ok, some are shifted from one or two pixels in axis x and y
> What can be modified in ucg_dev_ic_st7735.c, ucg_dev_tft_128x160_st7735.c to change the offset?
> Thanks
> 
> [image]

Hi, I have the same issue like yours. Where can I download your "offset_test1_st773" ? can you send me the link to the program. Thanks

karawin, 2020-03-31:
Sources are here
https://github.com/karawin/Ka-Radio32/tree/master/components/ucglib
Integrated in the standat delivery.

lucas-mercier, 2021-05-31:
Hi,
Same problem here, and can't really figure out how to use your source. The files can't be added to my project, as a library, because it's not "Make".
How can I add your library ?

arduware, 2022-04-08:
Hallo - super. Funktioniert bei mir auch. DANKE