olikraus/ucglib #114: ILI 9486 320x480 SPI not working

cchandel, 2019-04-22:
Hi,
   I'm trying to use a 320x480 tft display from [elecrow] (https://www.elecrow.com/wiki/index.php?title=3.5_Inch_480x320_TFT_Display_with_Touch_Screen_for_Raspberry_Pi) connected over SPI with a BluePill (STM32F103)

   The display remains white and flickers - but I can't see any graphics on the display while running the GraphicsTest example. The init code from the rasp pi .dtb is :-

> init = <0x10000b0 0x0 0x1000011 0x20000ff 0x100003a 0x55 0x1000036 0x28 0x10000c2 0x44 0x10000c5 0x0 0x0 0x0 0x0 0x10000e0 0xf 0x1f 0x1c 0xc 0xf 0x8 0x48 0x98 0x37 0xa 0x13 0x4 0x11 0xd 0x0 0x10000e1 0xf 0x32 0x2e 0xb 0xd 0x5 0x47 0x75 0x37 0x6 0x10 0x3 0x24 0x20 0x0 0x10000e2 0xf 0x32 0x2e 0xb 0xd 0x5 0x47 0x75 0x37 0x6 0x10 0x3 0x24 0x20 0x0 0x1000036 0x28 0x1000011 0x1000029>

   Any ideas?

Hancock888, 2019-04-29:
Hi not working!!!! help

cchandel, 2019-04-30:
Hi,
      In the interim, I've used https://github.com/jaretburkett/ILI9488.git to drive my 9486 display.
      Surprisingly, the display works better with this library.
      On the pi, the same display is identified as a 9486 - so there's something the 9488 drivers are doing that's working. I think it could be the init code - but haven't gotten around to playing with that.

      Hope this helps.
Thanks.

gioreva, 2019-08-05:
Same problem.
Help.
Has anyone fixed the register for ILI9488?

gioreva, 2019-08-05:
This work well wit ILI9488, unique problem, text are mirrored.

```
static const ucg_pgm_uint8_t ucg_tft_320x480_ili9486_init_seq[] = {
    UCG_CFG_CD(0, 1), /* DC=0 for command mode, DC=1 for data and args */
    UCG_RST(1),
    UCG_CS(1), /* disable chip */
    UCG_DLY_MS(1),
    UCG_RST(0),
    UCG_DLY_MS(50),
    UCG_RST(1),
    UCG_DLY_MS(100),
    UCG_CS(0), /* enable chip */

    // UCG_C19(0XF2, 0x18, 0xA3, 0x12, 0x02, 0XB2, 0x12, 0xFF, 0x10, 0x00),
    // UCG_C12(0XF8, 0x21, 0x04),
    // UCG_C10(0x13),             // Normal Display Mode ON
    // UCG_C11(0x36, 0x08),       // Memory Access Control, BGR Order
    // UCG_C11(0xB4, 0x02),       // Display Inversion Control, 2-dot inversion
    // UCG_C12(0xB6, 0x02, 0x22), // Display Function Control, AGND, Normal scan
    // UCG_C11(0xC1, 0x41),       // Power Control 2
    // UCG_C12(0xC5, 0x00, 0x18), // VCOM Control 1
    // UCG_C11(0x3a, 0x66),       // Interface Pixel Format, 18bit
    // UCG_DLY_MS(50),
	
    UCG_C1F(0xE0,0x00,0x03,0x09,0x08,0x16,0x0A,0x3F,0x78,0x4C,0x09,0x0A,0x08,0x16,0x1A,0x0F),  // Positive Gamma Control
    UCG_C1F(0XE1,0x00,0x16,0x19,0x03,0x0F,0x05,0x32,0x45,0x46,0x04,0x0E,0x0D,0x35,0x37,0x0F),  // Negative Gamma Control
    UCG_C12(0XC0,0x17,0x15), // Power Control 1
	UCG_C11(0xC1,0x41), // Power Control 2
	UCG_C13(0xC5,0x00,0x12,0x80),// VCOM Control
//    writecommand(TFT_MADCTL); // Memory Access Control
    //writedata(0x48);          // MX, BGR
    UCG_C11(0x3A,0x66),  // 18 bit colour for SPI
    UCG_C11(0xB0,0x00),	// Interface Mode Control
    UCG_C11(0xB1,0xA0),// Frame Rate Control
    UCG_C11(0xB4,0x02),// Display Inversion Control
    UCG_C13(0xB6,0x02,0x02,0x3B),// Display Function Control
    UCG_C11(0xB7,0xC6),// Entry Mode Set
    UCG_C14(0xF7,0xA9,0x51,0x2C,0x82),// Adjust Control 3

//------------------------	

    UCG_C10(0x11), // Sleep OUT
    UCG_DLY_MS(120),
    UCG_C10(0x29), // Display ON
    UCG_C10(0x2C), // Memory Write

    UCG_CS(1), /* disable chip */
    UCG_END(), /* end of sequence */
};
```

gioreva, 2019-08-05:
Ok, Now work well.
Can you please add it at the library ?
I have taken reference from UTFT

`static const ucg_pgm_uint8_t ucg_tft_320x480_ili9486_init_seq[] = {
    UCG_CFG_CD(0, 1), /* DC=0 for command mode, DC=1 for data and args */
    UCG_RST(1),
    UCG_CS(1), /* disable chip */
    UCG_DLY_MS(1),
    UCG_RST(0),
    UCG_DLY_MS(50),
    UCG_RST(1),
    UCG_DLY_MS(100),
    UCG_CS(0), /* enable chip */

    UCG_C1F(0xE0,0x00,0x03,0x09,0x08,0x16,0x0A,0x3F,0x78,0x4C,0x09,0x0A,0x08,0x16,0x1A,0x0F),  // Positive Gamma Control
    UCG_C1F(0XE1,0x00,0x16,0x19,0x03,0x0F,0x05,0x32,0x45,0x46,0x04,0x0E,0x0D,0x35,0x37,0x0F),  // Negative Gamma Control
    UCG_C12(0XC0,0x17,0x15), // Power Control 1
	UCG_C11(0xC1,0x41), // Power Control 2
	UCG_C13(0xC5,0x00,0x12,0x80),// VCOM Control
    UCG_C11(0x36, 0x48),       // Memory Access Control, BGR Order
    UCG_C11(0x3A,0x66),  // 18 bit colour for SPI
    UCG_C11(0xB0,0x00),	// Interface Mode Control
    UCG_C11(0xB1,0xA0),// Frame Rate Control
    UCG_C11(0xB4,0x02),// Display Inversion Control
    UCG_C13(0xB6,0x02,0x02,0x3B),// Display Function Control
    UCG_C11(0xB7,0xC6),// Entry Mode Set
    UCG_C14(0xF7,0xA9,0x51,0x2C,0x82),// Adjust Control 3
	
	UCG_C12(0xB6, 0x02, 0x22), // Display Function Control, AGND, Normal scan

    UCG_C10(0x11), // Sleep OUT
    UCG_DLY_MS(120),
    UCG_C10(0x29), // Display ON
    UCG_C10(0x2C), // Memory Write

    UCG_CS(1), /* disable chip */
    UCG_END(), /* end of sequence */
};`

gioreva, 2020-02-12:
My change work well on ILI9488 controller.
Can please add it ?

cancian123, 2022-12-09:
Hi guys. I just testes giovera's solution and it worked fine. Can anyone update the library with the code that fixes the problem?