olikraus/ucglib #115: Does drawXBMP function available in ucglib?

venkatesh223, 2019-04-24:
Does ucglib support drawxbmp like the U8g2 library?
If no can you help me with an alternative on how to display an image using ucglib arduino please.

AnHardt, 2019-05-07:
(For me) it doesn't make much sense to implement a routine for displaying a uncompressed monochrome image format in a library for color displays.
Better use a library to decode your preferred image format and draw pixel by pixel.

ditor2, 2019-06-28:
I would like to press on this issue.
This does make sense when you would like to use a color display (and ucg) as a drop-in replacement for an existing monochrome-displayed gadget with a large set of xbm pictures. Most of the time it is enough to display xbm as a black and white figure without redesigning/redrawing it.

hznupeter, 2019-08-16:
use the setcolor and drawpixel to draw each point. you need two for loops.
ucg_SetColor(ucg, 0, 255, 255, 255);
ucg_DrawPixel(ucg, 50, 20);

Devilscave, 2021-10-30:
I open a Issues Mounth before -> #98 whit no answere.
@hznupeter that is a way you can go, but whit a picture whit hundrets of Pixels you need mutch time to greate this. 
Do you know a program which can determine the pixel position based on the RAW image data or the C data?
That would help.