olikraus/ucglib #118: cannot print degree ° symbol

rotarucosminleonard, 2019-08-03:
I cannot print the degree symbol on my st7739 display. It just does not show up.

iltis42, 2019-10-22:
Did you check if your font does support this ? 
ucg.print("\0xb0")  should do that, no problem i see as long the font supports. 
What font exactly did you try ?

Sub-7, 2019-12-14:
the same problem!
tested with ("°" also with "\0xb0")
Display: 1.77 TFT 160x128
Fonds:
ucg.setFont(ucg_font_7x13_mr);
ucg.setFont(ucg_font_courR12_mr);
ucg.setFont(ucg_font_courR10_mr);
ucg.setFont(ucg_font_profont10_mr);

Which font works?

kevinstadler, 2020-04-24:
I think this might be an issue with Arduino string encoding rather than with ucglib. The following works for me (note: no quotes around it, this is the raw byte notation of the ASCII character!)
```
ucg.drawGlyph(x, y, 0, 0xb0); // degree sign
```