olikraus/ucglib #121: Fix build warnings

fornellas, 2019-11-21:
I'm getting a fair amount of warnings when building the library. They should all be simple to fix. @olikraus  would you be willing to merge a PR if I work on it?

```
  CC	../ucglib/csrc/ucg_rotate.c
../ucglib/csrc/ucg_rotate.c:68:11: warning: no previous prototype for 'ucg_dev_rotate90' [-Wmissing-prototypes]
 ucg_int_t ucg_dev_rotate90(ucg_t *ucg, ucg_int_t msg, void *data)
           ^~~~~~~~~~~~~~~~
../ucglib/csrc/ucg_rotate.c:149:11: warning: no previous prototype for 'ucg_dev_rotate180' [-Wmissing-prototypes]
 ucg_int_t ucg_dev_rotate180(ucg_t *ucg, ucg_int_t msg, void *data)
           ^~~~~~~~~~~~~~~~~
../ucglib/csrc/ucg_rotate.c:208:11: warning: no previous prototype for 'ucg_dev_rotate270' [-Wmissing-prototypes]
 ucg_int_t ucg_dev_rotate270(ucg_t *ucg, ucg_int_t msg, void *data)
           ^~~~~~~~~~~~~~~~~
  CC	../ucglib/csrc/ucg_com_msg_api.c
../ucglib/csrc/ucg_com_msg_api.c:38:9: warning: no previous prototype for 'ucg_com_template_cb' [-Wmissing-prototypes]
 int16_t ucg_com_template_cb(ucg_t *ucg, int16_t msg, uint32_t arg, uint8_t *data)
         ^~~~~~~~~~~~~~~~~~~
../ucglib/csrc/ucg_com_msg_api.c: In function 'ucg_com_template_cb':
../ucglib/csrc/ucg_com_msg_api.c:38:36: warning: unused parameter 'ucg' [-Wunused-parameter]
 int16_t ucg_com_template_cb(ucg_t *ucg, int16_t msg, uint32_t arg, uint8_t *data)
                                    ^~~
../ucglib/csrc/ucg_com_msg_api.c:38:63: warning: unused parameter 'arg' [-Wunused-parameter]
 int16_t ucg_com_template_cb(ucg_t *ucg, int16_t msg, uint32_t arg, uint8_t *data)
                                                               ^~~
../ucglib/csrc/ucg_com_msg_api.c:38:77: warning: unused parameter 'data' [-Wunused-parameter]
 int16_t ucg_com_template_cb(ucg_t *ucg, int16_t msg, uint32_t arg, uint8_t *data)
                                                                             ^~~~
../ucglib/csrc/ucg_com_msg_api.c: At top level:
../ucglib/csrc/ucg_com_msg_api.c:182:6: warning: no previous prototype for 'ucg_com_SendStringP' [-Wmissing-prototypes]
 void ucg_com_SendStringP(ucg_t *ucg, uint16_t cnt, const ucg_pgm_uint8_t *byte_ptr)
      ^~~~~~~~~~~~~~~~~~~
  CC	../ucglib/csrc/ucg_dev_ic_hx8352c.c
../ucglib/csrc/ucg_dev_ic_hx8352c.c:140:11: warning: no previous prototype for 'ucg_handle_hx8352c_l90fx' [-Wmissing-prototypes]
 ucg_int_t ucg_handle_hx8352c_l90fx(ucg_t *ucg)
           ^~~~~~~~~~~~~~~~~~~~~~~~
../ucglib/csrc/ucg_dev_ic_hx8352c.c:180:11: warning: no previous prototype for 'ucg_handle_hx8352c_l90tc' [-Wmissing-prototypes]
 ucg_int_t ucg_handle_hx8352c_l90tc(ucg_t *ucg)
           ^~~~~~~~~~~~~~~~~~~~~~~~
../ucglib/csrc/ucg_dev_ic_hx8352c.c:278:11: warning: no previous prototype for 'ucg_handle_hx8352c_l90se' [-Wmissing-prototypes]
 ucg_int_t ucg_handle_hx8352c_l90se(ucg_t *ucg)
           ^~~~~~~~~~~~~~~~~~~~~~~~
../ucglib/csrc/ucg_dev_ic_hx8352c.c: In function 'ucg_handle_hx8352c_l90se':
../ucglib/csrc/ucg_dev_ic_hx8352c.c:306:15: warning: declaration of 'i' shadows a previous local [-Wshadow]
     ucg_int_t i, j;
               ^
../ucglib/csrc/ucg_dev_ic_hx8352c.c:280:11: note: shadowed declaration is here
   uint8_t i;
           ^
../ucglib/csrc/ucg_dev_ic_hx8352c.c:306:18: warning: variable 'j' set but not used [-Wunused-but-set-variable]
     ucg_int_t i, j;
                  ^
../ucglib/csrc/ucg_dev_ic_hx8352c.c: In function 'ucg_ext_hx8352c_18':
../ucglib/csrc/ucg_dev_ic_hx8352c.c:404:63: warning: unused parameter 'data' [-Wunused-parameter]
 ucg_int_t ucg_ext_hx8352c_18(ucg_t *ucg, ucg_int_t msg, void *data)
                                                               ^~~~
  CC	../ucglib/csrc/ucg_init.c
../ucglib/csrc/ucg_init.c:45:6: warning: no previous prototype for 'ucg_init_struct' [-Wmissing-prototypes]
 void ucg_init_struct(ucg_t *ucg)
      ^~~~~~~~~~~~~~~
  CC	../ucglib/csrc/ucg_dev_ic_pcf8833.c
../ucglib/csrc/ucg_dev_ic_pcf8833.c:149:11: warning: no previous prototype for 'ucg_handle_pcf8833_l90fx' [-Wmissing-prototypes]
 ucg_int_t ucg_handle_pcf8833_l90fx(ucg_t *ucg)
           ^~~~~~~~~~~~~~~~~~~~~~~~
../ucglib/csrc/ucg_dev_ic_pcf8833.c:192:11: warning: no previous prototype for 'ucg_handle_pcf8833_l90tc' [-Wmissing-prototypes]
 ucg_int_t ucg_handle_pcf8833_l90tc(ucg_t *ucg)
           ^~~~~~~~~~~~~~~~~~~~~~~~
../ucglib/csrc/ucg_dev_ic_pcf8833.c:277:11: warning: no previous prototype for 'ucg_handle_pcf8833_l90se' [-Wmissing-prototypes]
 ucg_int_t ucg_handle_pcf8833_l90se(ucg_t *ucg)
           ^~~~~~~~~~~~~~~~~~~~~~~~
../ucglib/csrc/ucg_dev_ic_pcf8833.c: In function 'ucg_handle_pcf8833_l90se':
[... 296 lines cut ...]
../ucglib/csrc/ucg_dev_ic_ili9486.c:359:63: warning: unused parameter 'data' [-Wunused-parameter]
 ucg_int_t ucg_ext_ili9486_18(ucg_t *ucg, ucg_int_t msg, void *data)
                                                               ^~~~
  CC	../ucglib/csrc/ucg_dev_ic_ssd1331.c
../ucglib/csrc/ucg_dev_ic_ssd1331.c:63:11: warning: no previous prototype for 'ucg_handle_ssd1331_l90fx' [-Wmissing-prototypes]
 ucg_int_t ucg_handle_ssd1331_l90fx(ucg_t *ucg)
           ^~~~~~~~~~~~~~~~~~~~~~~~
../ucglib/csrc/ucg_dev_ic_ssd1331.c:122:11: warning: no previous prototype for 'ucg_handle_ssd1331_l90tc' [-Wmissing-prototypes]
 ucg_int_t ucg_handle_ssd1331_l90tc(ucg_t *ucg)
           ^~~~~~~~~~~~~~~~~~~~~~~~
../ucglib/csrc/ucg_dev_ic_ssd1331.c:231:11: warning: no previous prototype for 'ucg_handle_ssd1331_l90se' [-Wmissing-prototypes]
 ucg_int_t ucg_handle_ssd1331_l90se(ucg_t *ucg)
           ^~~~~~~~~~~~~~~~~~~~~~~~
../ucglib/csrc/ucg_dev_ic_ssd1331.c: In function 'ucg_handle_ssd1331_l90se':
../ucglib/csrc/ucg_dev_ic_ssd1331.c:260:15: warning: declaration of 'i' shadows a previous local [-Wshadow]
     ucg_int_t i, j;
               ^
../ucglib/csrc/ucg_dev_ic_ssd1331.c:233:11: note: shadowed declaration is here
   uint8_t i;
           ^
../ucglib/csrc/ucg_dev_ic_ssd1331.c: In function 'ucg_ext_ssd1331_18':
../ucglib/csrc/ucg_dev_ic_ssd1331.c:362:63: warning: unused parameter 'data' [-Wunused-parameter]
 ucg_int_t ucg_ext_ssd1331_18(ucg_t *ucg, ucg_int_t msg, void *data)
                                                               ^~~~
```

olikraus, 2019-11-23:
Actually I do not work on ucglib any more...

fornellas, 2019-11-24:
Oh, sorry about that. So, does this mean the project is stale now? Maybe I can help carry it on if you want. I like ucglib a lot. I'm using it in a new project, for which I'd love to implement frame buffer to ucglib, but I'd feel less inclined to do it there's no chance for merging upstream.

olikraus, 2019-11-26:
I would be happy if someone would continue here... However I am not sure how much I could contribute here.
At least I could test the old behavior...
And maybe I could also integrate the u8g2 font render engine... so that we can takeover all the latest fonts from u8g2.

fornellas, 2019-11-27:
That would be great! You do have experience on the problem domain, even small comments on pull requests are good.

I'll do this PR for fixing warnings for starter.

PR #123 (2020-01-01): Fix warnings
Closes #121 .
closed 2020-01-01 by olikraus