olikraus/ucglib #128: Please add ST7789VW support

iggymayer, 2020-02-23:
Hello, I have a 1.3-inch 240 * 240 screen with ST7789VW driver chip. Please add support for this in UCGLIB. Related documents are in the link.
https://mega.nz/#!MRl3GQRb!k2YHsE-QhVTFinsR2yEMOOTEs6ncpiJspijSWGNe5s8

Thank you!

olikraus, 2020-02-23:
I have stopped working on ucglib. Nevertheless, I am willing to accept pull requests.

iggymayer, 2020-02-23:
I am a beginner, can you tell me how to add a new screen in UCGLIB?

olikraus, 2020-02-23:
"beginner"? Where exactly? 
You probably need some programming skills to do this.

To add a new display, you need to create a device procedure for this display. Have a look here: https://github.com/olikraus/ucglib/blob/master/csrc/ucg_dev_tft_128x160_st7735.c

Basically you need to add code for each of the messages passed to that procedure and handle them correctly. Probably you can takeover most of the code from other devices.

Messages start with "UCG_MSG_DEV_xxx" and are listed here: https://github.com/olikraus/ucglib/blob/master/csrc/ucg.h#L454

A kind of a internal description is here: https://github.com/olikraus/ucglib/wiki/hal

The biggest problem from my perspective is the init sequence for the display (e.g. see here: https://github.com/olikraus/ucglib/blob/master/csrc/ucg_dev_tft_240x320_ssd1289.c#L40) . This is not a problem of the lib, but you need to figure out how to setup your display and your chip correctly.

iggymayer, 2020-02-23:
It's ... it's complicated ...=_=!

olikraus, 2020-02-23:
I know. That's why companies pay a lot of money for such libs.

iggymayer, 2020-02-26:
Thank you, I'll look for other types of screens.

closed 2020-02-27 by iggymayer