olikraus/ucglib #132: ST7789V 135x240 TTGO support?

BrickBee, 2020-05-02:
There is a nice little [ESP32 ](https://www.banggood.com/LILYGO-TTGO-T-Display-ESP32-CP2104-WiFi-bluetooth-Module-1_14-Inch-LCD-Development-Board-p-1522925.html) with a ST7789V display on top that's 135x240 even though the description says 240x320.
Can support for this be added to this fine library which is integrated with the NodeMCU Lua project?
According to the [related page](https://github.com/Xinyuan-LilyGO/TTGO-T-Display) on GitHub it should work with the regular ST7789 driver with a few tweaks. Color order is said to be BGR.

slorquet, 2020-07-01:
I have this screen without an esp32 and I am also interested.

fornellas, 2020-08-29:
@BrickBee  #142 should contain all code required, only difference being UCG_MSG_GET_DIMENSION.

charleslemaire0, 2022-12-27:
Hi Oli,

I would be happy If you could add this screen.
I can test it with 2 TTGO setup