olikraus/ucglib #150: ILI9342 driver

nbonniere, 2021-05-19:
I have been trying to modify the ILI9341 driver to make an ILI9342 driver but I am not successful so far. There is a blank area on the right when in landscape mode and an blank area at the top when rotated 90 deg.  Can't seem to find a register that would explain this offset.  Would you have time to generate a proper ILI9342 driver ?

slorquet, 2023-01-26:
Hi,

Where did you find this screen? in the m5stack? I have made a working driver for it. I'll send a pull request.

slorquet, 2023-01-26:
Well in fact I did it before and it is still not pushed upstream.

Is this what you need?

https://github.com/olikraus/Ucglib_Arduino/pull/23

nbonniere, 2023-01-27:
I updated my fork with my code. Match is very close, except for UCG_VARX and UCG_VARY. Not sure which is right (?)
https://github.com/nbonniere/ucglib

slorquet, 2023-01-27:
I like the ILI9342 name instead of my use of m5stack. I had doubt the ILI9342 was the chip name for the portrait screen but I see new docs now and it seem to be the right choice.

I had a look at the differences:
For UCG_VARX we do EXACTLY the same;
But for UCG_VARY:
```
 me:  UCG_C10(0x02b),	UCG_VARY(8,0x01, 0), UCG_VARY(0,0x0ff, 0), UCG_A2(0x000, 0x0ef),		/* set y position */
you:  UCG_C10(0x02b),	UCG_VARY(0,0x00, 0), UCG_VARY(0,0x0ff, 0), UCG_A2(0x000, 0x0ef),	
```
I have no idea what the differences are.

These are the definitions from ucg.h
/* access procedures to ucg->arg.pixel.pos.x und ucg->arg.pixel.pos.y */
```
#define UCG_VARX(s,a,o)				0x0a0 | ((s)&15), (a), (o)
#define UCG_VARY(s,a,o)				0x0b0 | ((s)&15), (a), (o)
```
I have used the same values in varx and vary by symmetry, what was your reason to use s=0 and a=0 instead?

nbonniere, 2023-01-27:
I reversed what was used in the 9341 since X is swapped with Y in the 9342. Pure guess at this time, and display works OK, so I didn't check further.
- 9341
  UCG_C10(0x02a),	UCG_VARX(0,0x00, 0), UCG_VARX(0,0x0ff, 0), UCG_A2(0x000, 0x0ef),	/* set x position */
  UCG_C10(0x02b),	UCG_VARY(8,0x01, 0), UCG_VARY(0,0x0ff, 0), UCG_A2(0x001, 0x03f),	/* set y position */
- 9342
  UCG_C10(0x02a),	UCG_VARX(8,0x01, 0), UCG_VARX(0,0x0ff, 0), UCG_A2(0x001, 0x03f),	/* set x position */
  UCG_C10(0x02b),	UCG_VARY(0,0x00, 0), UCG_VARY(0,0x0ff, 0), UCG_A2(0x000, 0x0ef),	/* set y position */

slorquet, 2023-01-27:
OK, makes sense, and thanks. I hope your driver gets integrated. It is frustrating that this project is not maintained anymore.